m-labs/nmigen #257: Unexpected behaviour of `Record.connect()`

akukulanski, 2019-10-22:
Hi,
We saw that `connect()` won't connect the input and output records, nevertheless the signals are connected if they are individually described. Is that an expected behaviour?

```python
class AxiStream(Record):
    def __init__(self, width, direction=None, name=None, fields=None):
        self.width = width
        if direction == 'sink':
            layout = [('TDATA', width, Direction.FANIN),
                      ('TVALID', 1, Direction.FANIN),
                      ('TREADY', 1, Direction.FANOUT),
                      ('TLAST', 1, Direction.FANIN)]
        elif direction == 'source':
            layout = [('TDATA', width, Direction.FANOUT),
                      ('TVALID', 1, Direction.FANOUT),
                      ('TREADY', 1, Direction.FANIN),
                      ('TLAST', 1, Direction.FANOUT)]
        Record.__init__(self, layout, name=name, fields=fields)

class DummyCore(Elaboratable):
    def __init__(self, width_in, width_out):
        self.input = AxiStream(width_in, 'sink', name='INPUT')
        self.output = AxiStream(width_out, 'source', name='OUTPUT')
    def elaborate(self, platform):
        m = Module()
        comb = m.d.comb
        #comb += self.output.connect(self.input) # does not connect the signals
        comb += self.output.TDATA.eq(self.input.TDATA)
        comb += self.output.TVALID.eq(self.input.TVALID)
        comb += self.output.TLAST.eq(self.input.TLAST)
        comb += self.input.TREADY.eq(self.output.TREADY)
        return m
```
Thanks,
@akukulanski, @andresdemski

whitequark, 2019-10-23:
The direction of `connect` goes the other way around: it's `m.d.comb += input.connect(output1, output2, ...)`

jordens, 2019-10-23:
This confuses me each time. Maybe there is a better name for `connect()`.

whitequark, 2019-10-23:
As I've said on IRC earlier, IMO we should deprecate `connect` and `Direction` entirely and rather have bus-specific interconnect modules.

akukulanski, 2019-10-23:
> The direction of `connect` goes the other way around: it's `m.d.comb += input.connect(output1, output2, ...)`

Inverting that doesn't make the difference. Here are the variants: https://github.com/akukulanski/nmigen_testcase

The confussion cames from the ambiguity that a module's port with *FANIN* (sink) has *FANOUT* (source) from inside the module. That's what makes that `.connect()` from inside a module has the opposite behaviour from what one would expect.

> As I've said on IRC earlier, IMO we should deprecate `connect` and `Direction` entirely and rather have bus-specific interconnect modules.

Agree!

sbourdeauducq, 2019-10-23:
> This confuses me each time.

I remember it by considering how it is used in the CSR bus with OR-multiplexing: ``initiator.connect(target1, target2, ...)``
But yes, it can be confusing.

closed 2019-10-28 by whitequark