m-labs/nmigen #260: Simple memory doesn't seem to work with a simple Cover statement?

RobertBaruch, 2019-10-27:
Possibly I'm not understanding how to use `Memory`?

mem.py:

```python
from nmigen import *
from nmigen.asserts import *
from nmigen.cli import main

class A(Elaboratable):
    def __init__(self):
        a = [1, 2, 3, 4, 5, 6]
        self.addr = Signal.range(0, len(a))
        self.data = Signal(8)
        self.mem = Memory(width=8, depth=len(a), init=a)

    def elaborate(self, platform):
        m = Module()
        m.submodules.rdport = rdport = self.mem.read_port()
        m.d.comb += [
            rdport.addr.eq(self.addr),
            self.data.eq(rdport.data),
        ]
        return m

if __name__ == "__main__":
    m = Module()
    a = A()
    m.submodules.a = a

    m.d.comb += Cover(a.addr == 0)

    main(m, ports=[a.addr, a.data])
```

mem.sby:

```
[tasks]
cover

[options]
cover: mode cover
cover: depth 61

[engines]
smtbmc boolector

[script]
read_ilang mem.il
prep -top top

[files]
mem.il
```

The output seems to imply that the memory is never initialized, and the output is always `10000000`.

whitequark, 2019-10-27:
That looks correct. Are you taking into account that memory ports (read and write) are synchronous by default?

RobertBaruch, 2019-10-28:
Ah, that was it. I had to set up the `read_port` properly:

```python
        m.submodules.rdport = rdport = self.mem.read_port(domain="comb")
```

Thanks!

closed 2019-10-28 by RobertBaruch