m-labs/nmigen #261: Const Record value?

RobertBaruch, 2019-10-28:
Suppose I have a `Layout` with various fields:

```python
class ThingLayout(Layout):
    def __init__(self):
        super().__init__([
            ("field1", unsigned(8)),
            ("field2", unsigned(16)),
            ...
        ])
```

Now I'd like to construct a Python integer for a specific setting for the fields in the `Layout`. I know it needs to be 24 bits. How do I construct such a value?

The ultimate goal is to initialize a `Memory` with lots of these values, then be able to read the memory and load a `Record` with the value from memory.

whitequark, 2019-10-28:
There's no such functionality currently; `Record`s can only be used in HDL. If such functionality existed, it would probably be present as a pair of functions, `Layout.pack` and `Layout.unpack`, which would turn a dictionary of ints into an int and vice versa.

RobertBaruch, 2019-10-28:
That sounds useful. As it is now, I'll just make a class that can pack and generate unpack statements during elaborate time.

RobertBaruch, 2019-10-28:
Basically something like this:

```python
class StepStruct(Object):
    def __init__(self, (step, arg1, arg2, next_cycle, next_step)):
        self.step = Const(step)
        self.arg1 = Const(arg1)
        self.arg2 = Const(arg2)
        self.next_cycle = Const(next_cycle)
        self.next_step = Const(next_step, shape=unsigned(10))

    def _as_const(self):
        return Cat(self.step, self.arg1, self.arg2, self.next_cycle,
                   self.next_step)._as_const()

    def uncat(self, m, value, (step, arg1, arg2, next_cycle, next_step)):
        m.d.comb += Cat(step, arg1, arg2, next_cycle, next_step).eq(value)
```

whitequark, 2019-10-28:
I don't understand what's that supposed to do. Also, `_as_const` is an internal API that may be removed at any time.

RobertBaruch, 2019-10-28:
Can you keep `_as_const()`? Otherwise how do I get the const value of a `Cat` of `Consts`?

whitequark, 2019-10-28:
Just like any other function starting with `_`, it may be changed or removed at any time. Why do you need to get it?

whitequark, 2019-10-28:
> Just like any other function starting with `_`, it may be changed or removed at any time.

Also, to be clear—if `_as_const` is indeed so useful that it needs to be kept, then it should be promoted to a stable public API instead. That's a separate question though, and if that would be to be done, it would be necessary to greatly expand it in scope to support every nMigen AST class and not just two of them at random.

RobertBaruch, 2019-10-28:
Suppose I have a bunch of structured values that I want to put into a memory. Memories can only store values of a given width. Therefore, I need to convert my structured values into a single value of fixed width. Now, I can do that using something like this:

```python
class StepStruct(Object):
    def __init__(self, (step, arg1, arg2, next_cycle, next_step)):
        self.step = Const(step)
        self.arg1 = Const(arg1)
        self.arg2 = Const(arg2)
        self.next_cycle = Const(next_cycle)
        self.next_step = Const(next_step, shape=unsigned(10))
        self.parts = [self.step, self.arg1, self.arg2, self.next_cycle, self.next_step]

    def value(self):
        value = 0
        for part in reversed(self.parts):
            value <<= len(part)
            value |= part.value
        return value
```

Ok, `Cat` actually does that for me, but now I have to duplicate the logic because `_as_const()` is internal and not to be used. In any case, now I can do something like this:

```python
    a = [(Step.X, Arg.A, Arg.B, Cycle.C, 10),
         (Step.Y, Arg.A, Arg.Z, Cycle.D, 20),
         ...
    ]

    mem_init_val = [StepStruct(v).value() for v in a]
    # Now I can finally initialize the memory
```

Next, I need to take a value out of memory and break it up. That would be very convenient with a `Cat` also! I could use it easily... but since `_as_const()` is internal, I have no way of ensuring that the logic I copied above is the same as `Cat` uses, which means I can't use `Cat` to do something like this:

```python
    m.d.comb += Cat(step, a1, a2, nc, ns).eq(mem[addr])
```
 
So what should I do instead?

whitequark, 2019-10-28:
Isn't `StepStruct` basically a `Record`? And `Record`s have a `Layout`, which you can interrogate to find out which part lives at which offset.

RobertBaruch, 2019-10-28:
Hmm, I see where you can get the shape of a `Layout`'s field, but not the offset.

whitequark, 2019-10-28:
> Hmm, I see where you can get the shape of a `Layout`'s field, but not the offset.

The fields are sequential and without gaps, so you can just sum the widths while you're iterating through the fields.

RobertBaruch, 2019-10-28:
Hmm.... I don't see anything in the code that makes that obvious, so I can't really rely on that. Where in the code is that implemented?

whitequark, 2019-10-28:
> Hmm.... I don't see anything in the code that makes that obvious, so I can't really rely on that.

This is the basic contract of `Record`: a `Record` always acts exactly like a `Cat` of its fields. Now that you mention it, this is not documented, which is an unfortunate omission. But it's a guarantee it provides.

RobertBaruch, 2019-10-28:
OK, just so I make sure I understand: that means the layout of 

```python
Layout([
    ("first", unsigned(8)),
    ("second", unsigned(16))
])
```

is the same as `Cat(first, second)`, so the low 8 bits is `first` and the next highest 16 bits is `second`. Does that sound right?

whitequark, 2019-10-28:
Correct. (In fact, a `Record` could be implemented as a `UserValue` that lowers to `Cat`--it predates `UserValue` so it doesn't, but it doesn't need to be special-cased anymore!)

RobertBaruch, 2019-10-28:
Sounds good, thanks!

closed 2019-10-28 by RobertBaruch