m-labs/nmigen #271: Constant sign-extends incorrectly.

kbob, 2019-11-17:
[This program](https://gist.github.com/kbob/9f94547498ec091d9f8412aaa156792d) shows that subtracting 1 from a signed signal actually adds one.

The generated Verilog shows this statement.  I think Yosys sign-extends `$signed(1'h1)` to `24'b111111111111111111111111` before subtracting.

      assign \$2  = $signed(bar) - (* src = "foo.py:14" *) $signed(1'h1);

Versions:

nMigen v0.1

Yosys 0.9+932 (git sha1 ab0fb19c, clang 11.0.0 -fPIC -Os)

whitequark, 2019-11-18:
Does it work correctly if you simulate the design?

kbob, 2019-11-18:
Yes.  pysim subtracts one.

whitequark, 2019-11-18:
Since both the Verilog and simulation behavior is correct, this isn't an nMigen bug.

programmerjake, 2019-11-18:
The generated Verilog is incorrect since a 1 bit signed value can only have the values -1 and 0. nmigen should zero-extend unsigned values when converting them to 
a wider signed type. The integer type conversion is needed since Verilog doesn't have (as far as I know) operations that have one signed and one unsigned operand.

See also: [C++ example on Compiler Explorer](https://gcc.godbolt.org/z/vQM4j8).

whitequark, 2019-11-18:
Ah, hm. You're right, I missed that it was 1-bit. That's troublesome.

closed 2019-11-18 by whitequark

kbob, 2019-11-18:
I confirm that the fix works for my code.  Thank you.

commit 7c322e56 (2019-11-18): back.rtlil: extend shorter operand of a binop when matching sign.

This is necessary because converting a large unsigned value to
a signed value of the same width may change its sign.

Fixes #271.