m-labs/nmigen #280: pysim2 doesn't write input signals to VCD

adamgreig, 2019-12-04:
Since updating to pysim2, any signals not driven from inside the fragment being simulated are not written to the VCD. This means signals which are inputs to the fragment and are driven by a sync process added to the simulator do not show up.

I think this is because FragmentCompiler is the only thing that adds signals to `signal_names`, but it is only called for the fragment the simulator is created with, not for any processes that are then added to the simulator. I might be misunderstanding the code though.

Additionally, as far as I can tell the `traces` argument to `write_vcd` is totally unused: it gets passed to `_VCDWaveformWriter.__init__`, which does nothing with it.

whitequark, 2019-12-04:
> I think this is because FragmentCompiler is the only thing that adds signals to `signal_names`, but it is only called for the fragment the simulator is created with, not for any processes that are then added to the simulator. I might be misunderstanding the code though.

This is basically correct, but what it should do instead is also add any inputs. This wouldn't solve signals used for communicating between simulation processes only, but your suggestion wouldn't solve it as well because you can't add new signals in the middle of a vcd file.

> Additionally, as far as I can tell the `traces` argument to `write_vcd` is totally unused: it gets passed to `_VCDWaveformWriter.__init__`, which does nothing with it.

Oops, good catch.

phlipped, 2019-12-10:
Would it be possible to just re-write the vcd file after simulation has finished such that the variable definitions section at the start contains any additional signals that were seen along the way?

whitequark, 2019-12-10:
> Would it be possible to just re-write the vcd file after simulation has finished such that the variable definitions section at the start contains any additional signals that were seen along the way?

Maybe, but I'm not really interested in any solution that requires me to manually manipulate VCD files.

JarrettBillingsley, 2019-12-20:
Seems like the VCDWriter is a bit inflexible in this regard... I can imagine a version which buffers the variable changes and registered variables in memory until it's closed, at which point it *then* writes everything out. Not sure if that could be done with subclassing...

adamgreig, 2019-12-20:
I'm not really fussed about outputting signals that testbenches might create (at least, it doesn't affect me), but not having any input signals at all in the VCD makes my usual debugging/testing with pysim2 basically impossible at the moment.

commit 882fddfa (2020-03-14): back.pysim: emit toplevel inputs in VCD files as well.

Before this commit, only signals driven from fragments (in practice,
everything except toplevel inputs) would get written to a VCD file.
Not having toplevel inputs in the dump made debugging ~impossible.

After this commit, all signals the fragment refers to get written to
a VCD file. (More specifically, all signals the compiler assigns
an index to, i.e. signals the generated code reads or writes.)

Fixes #280.
closed 2020-03-14 by sbourdeauducq