m-labs/nmigen #283: Record.like() should be a classmethod, not staticmethod.

kbob, 2019-12-20:
The comments explicitly say that Record can be subclassed, so it would be good for Record.like to clone a Record subclass instead of downclassing to Record.

nMigen version: v.01.

whitequark, 2019-12-21:
That's not possible: if you override `__init__` while subclassing `Record` and change the arguments in an incompatible way, then calling `Record.like` would fail.

closed 2019-12-21 by whitequark

Fatsie, 2019-12-21:
BTW inside #266 this was discussed. There the following doc for like was suggested:
```python
class Record(Value):
    @classmethod
    def like(cls, other, *, name=None, name_suffix=None, src_loc_at=0):
        """Return fresh object with same fields as other object

        Use Record.like(other) if one wants a Record object with same fields as other.
        Use other.__class__.like(other) if one wants an object of same class as other
        and with same fields."""
```
This way I think you can do what you want to do with current code.

kbob, 2019-12-21:
Thanks for the workaround.  That looks good.