m-labs/nmigen #286: Misleading error when trying to use self.comb when inheriting from Module

AsuMagic, 2019-12-26:
Consider the following migen code:
```python
class ORGate(Module):
	def __init__(self):
		self.a = Signal()
		self.b = Signal()
		self.x = Signal()

		self.comb += self.x.eq(self.a | self.b)
```

While this is not the valid way to implement this in nmigen, if you try to use this code as-is in nmigen, the error is misleading:

```
  File "/usr/lib/python3.8/site-packages/nmigen/hdl/dsl.py", line 75, in __getattr__
    raise AttributeError("'{}' object has no attribute '{}'; did you mean 'd.{}'?"
AttributeError: 'ORGate' object has no attribute 'comb'; did you mean 'd.comb'?
```

If my understanding is right, the direct alternative to the above code in nmigen is to inherit from `Elaboratable` and use the `elaborate()` mechanism.  
In that case, the error check should hint at this accordingly, instead of incorrectly hinting to use `d.comb`.

whitequark, 2020-02-01:
Fixed [upstream](https://github.com/nmigen/nmigen).

closed 2020-02-02 by sbourdeauducq

sbourdeauducq, 2020-02-02:
Merged [upstream](https://github.com/m-labs/nmigen).

commit 6fd7cbad (2020-02-02): hdl.dsl: don't allow inheriting from Module.

`Module` is an object with a lot of complex and sometimes fragile
behavior that overrides Python attribute accessors and so on.
To prevent user designs from breaking when it is changed, it is not
supposed to be inherited from (unlike in Migen), but rather returned
from the elaborate() method. This commit makes sure it will not be
inherited from by accident (most likely by users familiar with
Migen).

Fixes #286.