m-labs/nmigen #293: (Verilog) nMigen should provide better information about original source code locations

Maykeye, 2020-01-04:
Consider for example this report from yosys 

        Unreached cover statement at abc.sv:2402.

If I go there, I'll see

     always @* if (\$cover$en ) cover(\$cover$check );

and I'll have to jump to `\$cover$check`, maybe even  further to variables like `\$157`.

Instead nMigen should generate 
   
     `line 123 /mnt/e/nmigen/abc.py
     always @* if (\$cover$en ) cover(\$cover$check );

In other words

a) source location should be generated next to cover/assert statements, not just next to their corresponding variable 
b) Instead of comments like ` (* src = "/mnt/e/abc/cde.py:17" *)`, source locations should be generated with \`line so tools like SymbiYosys can point the right location (maybe optionally: verilog95 doesn't describe `line directive, it was added in verilog 2001)

whitequark, 2020-01-04:
If you are using SymbiYosys, do not emit Verilog from nMigen. Instead, emit RTLIL from nMigen, which already has far more precise location information than Verilog <code>\`line</code> directives can represent even in principle.

closed 2020-01-04 by whitequark

Maykeye, 2020-01-04:
Doesn't it only works with yosys and its friends? I had not only it in mind, but also vendor specific tools which support verilog and preprocessor directives.

whitequark, 2020-01-04:
> Doesn't it only works with yosys and its friends?

Correct.

> I had not only it in mind, but also vendor specific tools which support verilog and preprocessor directives.

There is little nMigen can do directly here, since it relies on Yosys to emit Verilog. It is possible to improve the Yosys Verilog backend to emit <code>\`line</code> directives, but I have no specific plans to do so.