m-labs/nmigen #297: Output enable for Intel platform on multi I/O pins

schwigi, 2020-01-06:
Not sure wether it is a bug or I have done something wrong.

I tried to get a 16 bit wide io pins, but the generated oe wire is only 1 bit. For the Intel altiobuf_bidir output enable signal must be as wide as the data lines.

Attached is a small python script to generate the output.

`
[ioTest.py.zip](https://github.com/m-labs/nmigen/files/4027867/ioTest.py.zip)
`

peteut, 2020-01-06:
@schwigi That looks like a bug to me,`oe` should be a vector:  [nmigen.vendor.intel](https://github.com/m-labs/nmigen/blob/master/nmigen/vendor/intel.py#L279). 

[image]

Refer to ALTIOBUF IP COre User Guide, UG-01024 for details.

schwigi, 2020-01-07:
I did some debugging runs and the problem is not in the platform.intel part.

In lib/io.py:43 the oe width is hard coded to 1. For Intel FPGAs it is clear that you must use width here as well. Not sure however how other FPGAs behave though.

whitequark, 2020-01-08:
> I did some debugging runs and the problem is not in the platform.intel part.

No, the problem is in `vendor.intel`.

> In lib/io.py:43 the oe width is hard coded to 1.

This is deliberate: a wide tristate signal is treated as a single entity for the purposes of output enable. That is how (n)Migen define its behavior, which is completely independent of any platform. Any platform that has output enable with the same width as output should use `Repl`.

If you want to control direction per-bit, you should instantiate a tristate per bit.

peteut, 2020-01-08:
@schwigi: shouldn't this https://github.com/m-labs/nmigen/blob/7ce75b85fc24c79ddf9bc18ff08678b739b0cd95/nmigen/vendor/intel.py#L288

also read `i_oe=self._get_oereg(m, pin),`?

schwigi, 2020-01-08:
@peteut: You are right, get_tristate should use `_get_oereg` as well.

closed 2020-01-09 by whitequark

ZirconiumX, 2020-01-10:
Thanks for correcting my oversight ^.^'

PR #300 (2020-01-09): Intel altiobuf_ requires oe with the same with as data lines.
Ah ok missunderstood the concept.

This should fix #297, with your suggestion.

Simple test looks fine on MAX10 FPGA.
commit 8184efd6 (2020-01-09): vendor.intel: fix output enable width for XDR=0 case.

Fixes #297.