m-labs/nmigen #301: vendor.xilinx_7series: Vivado TIMING-2 Warning

peteut, 2020-01-08:
*Vivado* issues TIMING-2 (default severity: CRITICAL) when `create_clock` uses a signal. 
Here is a potential solution to please `create_clock`: https://github.com/peteut/nmigen/commit/9e79a30655123c02a2a3025d8f5af0ce8c076b7f .

whitequark, 2020-01-09:
Can you explain what does the filter expression in your commit does?

peteut, 2020-01-09:
@whitequark this is indeed ugly and should be fixed, it's purpose is to filter negative ports of differential clock inputs, ending with `__n`. So it matches ports ending with `__p` or `__io`. Internal clocks are matched using `^.+/.+$`, for signals like `my_module/clk_out`.

whitequark, 2020-01-09:
You did explain why it's ugly, which is helpful (thanks!). Unfortunately, you didn't explain what it *does*, so I can't help you work out a nicer replacement. I don't have the Xilinx flavor of Tcl API memorized.

peteut, 2020-01-09:
Here's an excerpt from [UG906](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2019_2/ug906-vivado-design-analysis.pdf) explaining the TIMING-2 warning and how to use `create_clock` correctly:

[image]

The Tcl fragment does exactly that, it starts from the clock signal and determines it's startpoint using `all_fanin -flat -startpoints_only`. The returned could contains `my-signal__p my-signal__n`, therefore `filter -regexp` filters `my-signal__n`, so only a single item remain. 

```Tcl
[filter -regexp [all_fanin -flat -startpoints_only \
                    [get_nets {{signal|hierarchy("/")}}]] {NAME =~{(^.*__p$)|(^.*__io$)|(^.+/.+$)}}]
```

commit 3e2ecdf2 (2020-03-14): build.res,vendor: place clock constraint on port, not net, if possible.

For most toolchains, these are functionally identical, although ports
tend to work a bit better, being the common case. For Vivado, though,
it is necessary to place them on the port because its timing analyzer
considers input buffer delay.

Fixes #301.
closed 2020-03-14 by sbourdeauducq