m-labs/nmigen #303: Usability? Can we warn if a Statement is never assigned to a domain?

RobertBaruch, 2020-01-12:
All too often I've been charging ahead writing formal assertions, and I forget to put one in a domain, writing:

```
Assert(...)
```

instead of 

```
m.d.comb += Assert(...)
```

This sort of thing is silently ignored, so I go on merrily believing my formal verification worked, when in fact I simply never verified certain assertions.

If anything, this is especially nasty for Asserts, Assumes, and Covers that do not get added to a domain. Can there at least be a warning there? Is it possible?

whitequark, 2020-01-12:
Absolutely possible, and we should do this. We already do this for Elaboratables.

whitequark, 2020-02-01:
Fixed [upstream](https://github.com/nmigen/nmigen).

commit afece150 (2020-02-02): hdl.ast: warn on unused property statements (Assert, Assume, etc).

A property statement that is created but not added to a module is
virtually always a serious bug, since it can make formal verification
pass when it should not. Therefore, add a warning to it, similar to
UnusedElaboratable.

Doing this to all statements is possible, but many temporary ones are
created internally by nMigen, and the extensive changes required to
remove false positives are likely not worth the true positives.
We can revisit this in the future.

Fixes #303.
closed 2020-02-02 by sbourdeauducq