m-labs/nmigen #304: ResetSynchronizer clockdomain in submodule is not renamed properly with multiple submodule instances

povauboin, 2020-01-13:
```
Traceback (most recent call last):
  File "test.py", line 37, in <module>
    print(verilog.convert(top, ports=[top.clk, top.rst]))
  File "/mnt/data/prog/fpga/nmigen/nmigen/back/verilog.py", line 76, in convert
    rtlil_text = rtlil.convert(*args, **kwargs)
  File "/mnt/data/prog/fpga/nmigen/nmigen/back/rtlil.py", line 1010, in convert
    fragment = ir.Fragment.get(elaboratable, platform).prepare(**kwargs)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/ir.py", line 548, in prepare
    fragment = DomainLowerer()(fragment)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 315, in __call__
    return self.on_fragment(value)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 539, in on_fragment
    new_fragment = super().on_fragment(fragment)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 307, in on_fragment
    self.map_subfragments(fragment, new_fragment)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 270, in map_subfragments
    new_fragment.add_subfragment(self(subfragment), name)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 315, in __call__
    return self.on_fragment(value)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 539, in on_fragment
    new_fragment = super().on_fragment(fragment)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 307, in on_fragment
    self.map_subfragments(fragment, new_fragment)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 270, in map_subfragments
    new_fragment.add_subfragment(self(subfragment), name)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 315, in __call__
    return self.on_fragment(value)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 539, in on_fragment
    new_fragment = super().on_fragment(fragment)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 310, in on_fragment
    self.map_drivers(fragment, new_fragment)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 506, in map_drivers
    new_fragment.add_driver(self.on_value(signal), domain)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 106, in on_value
    new_value = self.on_ResetSignal(value)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 516, in on_ResetSignal
    domain = self._resolve(value.domain, value)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/xfrm.py", line 500, in _resolve
    raise DomainError("Signal {!r} refers to nonexistent domain '{}'"
nmigen.hdl.cd.DomainError: Signal (rst sub) refers to nonexistent domain 'sub'
```

MCVE:

```
from nmigen import *
from nmigen.back import verilog
from nmigen.lib.cdc import ResetSynchronizer

class MySubModule(Elaboratable):
    def __init__(self, clk, rst):
        self.clk = clk
        self.rst = rst

    def elaborate(self, platform):
        m = Module()

        m.domains.cd_sub = cd_sub = ClockDomain()
        m.d.comb += cd_sub.clk.eq(self.clk)
        m.submodules.reset_sync = ResetSynchronizer(self.rst, domain="sub")

        return m

class MyTopModule(Elaboratable):
    def __init__(self):
        self.clk = Signal()
        self.rst = Signal()
        self.counter = Signal(8)

    def elaborate(self, platform):
        m = Module()

        m.submodules.s1 = MySubModule(self.clk, self.rst)
        m.submodules.s2 = MySubModule(self.clk, self.rst)

        # m.d.s1_sub += self.counter.eq(self.counter + 1)

        return m

if __name__ == "__main__":
    top = MyTopModule()
    print(verilog.convert(top, ports=[top.clk, top.rst]))
```

povauboin, 2020-01-17:
Thanks @whitequark . Can you clarify how to use the renamed clock domains ?

Doing this:
```python
m.d.s1_sub += self.counter.eq(self.counter + 1)
```
yields the error:
```
Traceback (most recent call last):
  File "test.py", line 40, in <module>
    print(verilog.convert(top, ports=[top.clk, top.rst]))
  File "/mnt/data/prog/fpga/nmigen/nmigen/back/verilog.py", line 76, in convert
    rtlil_text = rtlil.convert(*args, **kwargs)
  File "/mnt/data/prog/fpga/nmigen/nmigen/back/rtlil.py", line 1010, in convert
    fragment = ir.Fragment.get(elaboratable, platform).prepare(**kwargs)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/ir.py", line 544, in prepare
    new_domains = fragment._propagate_domains(missing_domain)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/ir.py", line 394, in _propagate_domains
    self._propagate_domains_up()
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/ir.py", line 350, in _propagate_domains_up
    self.add_domains(domain)
  File "/mnt/data/prog/fpga/nmigen/nmigen/hdl/ir.py", line 145, in add_domains
    assert domain.name not in self.domains
AssertionError
```

But if the submodule is created with a DomainRenamer like this, it works:
```python
m.submodules.s1 = DomainRenamer({"sub": "s1_sub"})(
    MySubModule(self.clk, self.rst)                
)                                                  
```

whitequark, 2020-01-18:
> Can you clarify how to use the renamed clock domains ?

That's actually a different bug, one that is much harder to fix.

closed 2020-01-18 by whitequark

whitequark, 2020-01-18:
@povauboin Your example works now.

commit 7cb30953 (2020-01-17): hdl.xfrm: transform drivers as well in DomainRenamer.

This is necessary because drivers may be late bound.

Fixes #304.
commit a7be3b48 (2020-01-18): hdl.ir: resolve hierarchy conflicts before creating missing domains.

Otherwise, code such as:

    m.submodules.a = (something with cd_sync)
    m.submodules.b = (something with cd_sync)
    m.d.b_sync += x.eq(y)

causes an assertion failure.

Fixes #304 (again).