m-labs/nmigen #307: AssertionError domain.name not in self.domains

nicolas-robin, 2020-01-18:
Since a7be3b4, this code fails :

Code :
```
from nmigen import *
from nmigen_boards.ice40_hx8k_b_evn import ICE40HX8KBEVNPlatform

platform = ICE40HX8KBEVNPlatform()
top = Module()
top.d.sync += platform.request("led").o.eq(1)
platform.build(top)
```

Error :
```
Traceback (most recent call last):
  File "[...]/sandbox.py", line 7, in <module>
    platform.build(top)
  File "[...]/venv/lib/python3.7/site-packages/nmigen/build/plat.py", line 77, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "[...]/venv/lib/python3.7/site-packages/nmigen/build/plat.py", line 147, in prepare
    fragment = fragment.prepare(ports=self.iter_ports(), missing_domain=lambda name: None)
  File "[...]/venv/lib/python3.7/site-packages/nmigen/hdl/ir.py", line 556, in prepare
    new_domains = fragment._propagate_domains(missing_domain)
  File "[...]/venv/lib/python3.7/site-packages/nmigen/hdl/ir.py", line 403, in _propagate_domains
    self._propagate_domains_up()
  File "[...]/venv/lib/python3.7/site-packages/nmigen/hdl/ir.py", line 359, in _propagate_domains_up
    self.add_domains(domain)
  File "[...]/venv/lib/python3.7/site-packages/nmigen/hdl/ir.py", line 145, in add_domains
    assert domain.name not in self.domains
AssertionError
```

smunaut, 2020-01-20:
Same error happens when trying to run glasgow with latest nMigen.
(seen by me, @esden @miek at least)

smunaut, 2020-01-20:
Not sure if this is the correct fix, but this seems to work for me :

```diff
diff --git a/nmigen/hdl/ir.py b/nmigen/hdl/ir.py
index 26dec85..2b8b74c 100644
--- a/nmigen/hdl/ir.py
+++ b/nmigen/hdl/ir.py
@@ -142,7 +142,7 @@ class Fragment:
     def add_domains(self, *domains):
         for domain in flatten(domains):
             assert isinstance(domain, ClockDomain)
-            assert domain.name not in self.domains
+            assert (domain.name not in self.domains) or (self.domains[domain.name] == domain)
             self.domains[domain.name] = domain
 
     def iter_domains(self):

```

whitequark, 2020-01-20:
Likely not the correct fix, but an acceptable workaround.

whitequark, 2020-02-01:
Fixed [upstream](https://github.com/nmigen/nmigen).

commit cce6b868 (2020-02-02): build.plat: align pipeline with Fragment.prepare().

Since commit 7257c20a, platform code calls create_missing_domains()
before _propagate_domains_up() (as a part of prepare() call). Since
commit a7be3b48, without a platform, create_missing_domains() is
calle after _propagate_domains_up(); because of that, it adds
the missing domain to the fragment. When platform code then calls
prepare() again, this causes an assertion failure.

The true intent behind the platform code being written this way is
that it *overrides* a part of prepare()'s mechanism. Because it was
not changed when prepare() was  [... cut]
closed 2020-02-01 by nicolas-robin