m-labs/nmigen #311: Support for non integer connector / ressource 'numbers'

anuejn, 2020-01-23:
Currently nmigen.plat allows using integers to differentiate between different connectors / resources of the same type. In a design i am writing a platfrom for those are differentiated with cardinal points, which is quite helpful in this case.

whitequark, 2020-01-23:
The idea behind using numbers is that they are, well, sequential. In Migen, `platform.request` would pick the next lowest number. In nMigen, this is no longer the case (because getting different resources depending on the order in which you instantiate gateware is virtually always undesirable, if your gateware is composed of nested blocks, some of which require the same resource name).

There is still however some nMigen gateware that uses this sequential property: the built-in blinky. (I assume there could be user-developed gateware that e.g. enumerates SPI ports and creates a peripheral per port, too). I think this is sufficiently useful that we should in general keep doing that. Further, fortunately, even if you have e.g. "west" button, you can *also* have a "0th" button; you'll just be able to request it through one alias alone.

Therefore I think we can do this.

whitequark, 2020-01-31:
Fixed [upstream](https://github.com/nmigen/nmigen).

commit ec3a2193 (2020-02-02): build.dsl: allow strings to be used as connector numbers.

Fixes #311.
closed 2020-02-02 by sbourdeauducq