m-labs/nmigen #312: Assignment to a Record with zero-width fields generates invalid Verilog

jfng, 2020-01-24:
Repro:
```python3
from nmigen import *
from nmigen.back import rtlil, verilog

class Top(Elaboratable):
    def elaborate(self, platform):
        m = Module()

        rec = Record([
            ("a", 1),
            ("b", 0),
        ])

        m.d.comb += rec.eq(1)

        return m

if __name__ == "__main__":
    top = Top()
    print(verilog.convert(top))
    # print(rtlil.convert(top))
```

Verilog output:
```verilog
/* Generated by Yosys 0.9+1706 (git sha1 2bda51ac, clang 9.0.0 -fPIC -Os) */

(* generator = "nMigen" *)
(* top =  1  *)
(* \nmigen.hierarchy  = "top" *)
module top();
  (* src = "repro.py:9" *)
  wire rec__a;
  (* src = "repro.py:9" *)
  wire [-1:0] rec__b;
  assign "" = "";
  assign rec__a = 1'h1;
endmodule
```

RTLIL output:
```attribute \generator "nMigen"
attribute \top 1
attribute \nmigen.hierarchy "top"
module \top
  attribute \src "repro.py:9"
  wire width 1 \rec__a
  attribute \src "repro.py:9"
  wire width 0 \rec__b
  wire width 1 $verilog_initial_trigger
  process $group_0
    assign \rec__a 1'0
    assign \rec__b 0'0
    assign { \rec__b \rec__a } 1'1
    assign $verilog_initial_trigger $verilog_initial_trigger
    sync init
      update $verilog_initial_trigger 1'0
  end
  connect \rec__b 0'0
end
```

whitequark, 2020-01-24:
Hm, arguably this is a bug in Yosys, but I think we can just not generate assignments to zero width lvalues.

whitequark, 2020-01-31:
Fixed [upstream](https://github.com/nmigen/nmigen).

closed 2020-02-02 by sbourdeauducq

sbourdeauducq, 2020-02-02:
Fixed [upstream](https://github.com/m-labs/nmigen/).

commit 3ac13eb8 (2020-02-02): back.rtlil: don't emit wires for empty signals.

Fixes #312.