m-labs/nmigen #314: How to pass yosys_opts to LatticeICE40Platform?

RobertBaruch, 2020-01-26:
Specifically, I want to add `-abc2 -relut` to the yosys call in the generated `top.ys` file.

It's not clear from the comments or code how to do this. I see from the get_override() code in `plat.py` that you can get it from two places: an enviroment variable called `NMIGEN_yosys_opts`, or from `kwargs`, which doesn't seem to be defined:

https://github.com/m-labs/nmigen/blob/c42c3a096f89f8330f574ecd32700c2b9375e0be/nmigen/build/plat.py#L329

I'm probably missing some bit of Python knowledge, but I thought kwargs had to be passed in as an argument `**kwargs`.

In any case, I would much rather know how to pass `yosys_opts` programmatically, since environment variables aren't that great cross-platform-wise.

adamgreig, 2020-01-26:
The kwargs are passed in to the function that `get_override()` is defined inside. If you're using `LatticeICE40Platform` you can pass in `synth_opts` when you call `build()`: https://github.com/m-labs/nmigen/blob/c42c3a096f89f8330f574ecd32700c2b9375e0be/nmigen/vendor/lattice_ice40.py#L27

RobertBaruch, 2020-01-26:
Ah, I didn't notice that `get_override()` was defined in `toolchain_prepare()`. In any case, passing `synth_opt="-relut -abc2"` worked. Thanks!

closed 2020-01-26 by RobertBaruch