m-labs/nmigen #317: Using Python to Formal Verify Verilog and VHDL Files

goktug97, 2020-02-08:
As you know it would be hard to convert people from Verilog or VHDL but using Python to formal verify existing VHDL and Verilog codes would be awesome. I've used https://github.com/tgingold/ghdlsynth-beta and written formal verification codes in Verilog to verify VHDL codes. I thought using only formal verification part of the nMigen to generate that Verilog file.

AND Gate in VHDL
```vhdl
library ieee;
use ieee.std_logic_1164.all;
 
entity and_gate is
  port (
    a : in  std_logic;
    b : in  std_logic;
    y : out std_logic
    );
end and_gate;
 
architecture rtl of and_gate is
begin
  y <= a and b;
end rtl;
```

Verilog File for Formal Verification
```verilog
module simple_formal (
    input a,
    input b,
    output y
);

and_gate and1 (
    .a(a),
    .b(b),
    .y(y)
);

always @(*) begin
    assert(y == (a & b));
end

endmodule

```

The nMigen Python file is a little bit hacky to get what I want.
Python File
```py
class Dummy_AND(Elaboratable):
    def __init__(self):
        self.a = Signal()
        self.b = Signal()
        self.y = Signal()

    def elaborate(self, platform):
        m = Module()
        m.d.comb += self.y.eq(self.a + self.b)
        return m

    def ports(self):
        return [self.a, self.b, self.y]

if __name__ == '__main__':
    m = Module()
    m.submodules.and_gate = and_gate =  Dummy_AND()
    m.d.comb += Assert(and_gate.y == (and_gate.a & and_gate.b))

    with open('test.v', 'w') as f:
        found = False
        code = verilog.convert(m)
        code = code.split('\n')
        for line in code:
            if "module top" in line:
                found = True
            if found:
                f.write(f'{line}\n')
```
Generated Verilog File
```verilog
module top(b, a);
  (* src = "test.py:41" *)
  wire \$1 ;
  (* src = "test.py:41" *)
  wire \$3 ;
  (* src = "test.py:41" *)
  reg \$assert$check ;
  (* src = "test.py:41" *)
  reg \$assert$en ;
  (* src = "test.py:11" *)
  input a;
  (* src = "test.py:13" *)
  wire and_gate_y;
  (* src = "test.py:12" *)
  input b;
  assign \$1  = a & (* src = "test.py:41" *) b;
  assign \$3  = and_gate_y == (* src = "test.py:41" *) \$1 ;
  always @* if (\$assert$en ) assert(\$assert$check );
  and_gate and_gate (
    .a(a),
    .b(b),
    .y(and_gate_y)
  );
  always @* begin
    \$assert$en  = 1'h0;
    \$assert$check  = 1'h0;
    \$assert$check  = \$3 ;
    \$assert$en  = 1'h1;
  end
endmodule
```

The above solution works perfectly for this simple case. The problem is I need to add this

```python
m.d.comb += self.y.eq(self.a + self.b)
```
wrong statement to tell nMigen which ports are input and which ports are output. Also without this
it doesn't generate

```verilog
and_gate and_gate (
  .a(a),
  .b(b),
  .y(and_gate_y)
);
```
part. It feels a little hacky and it might fail in complex situations. Is there a better way to accomplish this ?

whitequark, 2020-02-08:
Instead of `Dummy_AND`, you should use an `Instance`, which lets you explicitly specify the ports.

goktug97, 2020-02-08:
Do you have an example. I've tried severtal things, for example

```python
    a = Signal()
    b = Signal()
    y = Signal()
    and_gate = Instance('and', i_a=a, i_b=b, o_y=y)
    m = Module()
    m.submodules.and_gate = and_gate
    m.d.comb += Assert(and_gate.ports.y == (and_gate.ports.a & and_gate.ports.b))
    m.d.comb += Cover(and_gate.porsts.y == 1)
```
but it failed
```
AttributeError: 'SignalDict' object has no attribute 'y'
```
Do you have an example ?

whitequark, 2020-02-08:
> Do you have an example

Your snippet is almost correct. Use:

```python
    a = Signal()
    b = Signal()
    y = Signal()
    and_gate = Instance('and', i_a=a, i_b=b, o_y=y)
    m = Module()
    m.submodules.and_gate = and_gate
    m.d.comb += Assert(y == (a & b))
    m.d.comb += Cover(y == 1)
```

goktug97, 2020-02-08:
Much Easier :) Thank you

closed 2020-02-08 by goktug97