m-labs/nmigen #319: Index a signal with a slice from another signal

rnd2, 2020-03-09:
Hi,

I am trying to convert an example from the TinyFPGA-BX repo ([source](https://github.com/tinyfpga/TinyFPGA-BX/blob/master/icestorm_template/top.v)) to nmigen.

```verilog
// look in pins.pcf for all the pin names on the TinyFPGA BX board
module top (
    input CLK,    // 16MHz clock
    output LED,   // User/boot LED next to power LED
    output USBPU  // USB pull-up resistor
);
    // drive USB pull-up resistor to '0' to disable USB
    assign USBPU = 0;

    ////////
    // make a simple blink circuit
    ////////

    // keep track of time and location in blink_pattern
    reg [25:0] blink_counter;

    // pattern that will be flashed over the LED over time
    wire [31:0] blink_pattern = 32'b101010001110111011100010101;

    // increment the blink_counter every clock
    always @(posedge CLK) begin
        blink_counter <= blink_counter + 1;
    end
    
    // light up the LED according to the pattern
    assign LED = blink_pattern[blink_counter[25:21]];
endmodule
```

Here is the nmigen based example:

```python
from nmigen import *
from nmigen_boards.tinyfpga_bx import TinyFPGABXPlatform

class Blinky(Elaboratable):
    def elaborate(self, platform):
        led   = platform.request("led", 0)
        counter = Signal(26)
        pattern = Signal(32, reset=0b101010001110111011100010101)

        m = Module()
        m.d.sync += counter.eq(counter + 1)
        m.d.comb += led.o.eq(pattern[counter[21:25]])
        return m

if __name__ == "__main__":
    platform = TinyFPGABXPlatform()
    platform.build(Blinky(), do_program=True)

```
The above code produces the following error: `TypeError: Cannot index value with (slice (sig counter) 21:25)`

I looked into nmigen source code and it seems that slices are not a valid key for other slices.
Does that mean is not supported at the moment or in general?

Is it possible for nmigen to generate the following
```verilog
assign LED = blink_pattern[blink_counter[25:21]];
```
Is there an alternative declaration (for nmigen) that will achieve the same result?

---

I did try `Array(pattern)[counter[21:25]]` which works, but generates a case statement.
Also, 
```python
m.d.comb += led.eq(pattern.bit_select(counter[21:25],1))
```
Which seems quite close to the original.

programmerjake, 2020-03-09:
Try using `Value.part` or one of the shift operators.

rnd2, 2020-03-09:
Thanks.
So, since `Value.part` is deprecated, the solution with `bit_select` on the previous post is one efficient way to do it.

commit 8947096e (2020-03-14): back.pysim: accept write_vcd(vcd_file=None).

Because write_vcd() is a context manager, this is useful if the VCD
file should be sometimes not written, since it avoids awkward
conditionals with duplicated code. It's not very elegant though.

Fixes #319.
closed 2020-03-14 by sbourdeauducq