m-labs/nmigen #320: Unable to Build and flash my board.

teezzan, 2020-03-12:
Hello,
 I just transitioned from migen to nmigen and I wrote a small code to test the VGA port of my de10-lite FPGA. The complete folder of the project can be found here (https://github.com/teezzan/vga-nmigen). The main part of the code is as follows.
```python

from nmigen import *
from vga_sync import *
from de10_lite import *

from nmigen.build import *
from nmigen.vendor.intel import *

platform = DE10LitePlatform()

class VGA_test(Elaboratable):
    
    def __init__(self):
        self.tv_out = Signal()
        self.x   = Signal()
        self.y   = Signal()
        self.vsync   = Signal()
        self.hsync  = Signal()
    def elaborate(self):
        m = Module()
        vga_tem = VGA_sync()
        m.submodules += vga_tem        
        vga = platform.request("vga")
        m.d.comb += [
            vga.vsync_n.eq(~vga_tem.vsync),
            vga.hsync_n.eq(~vga_tem.hsync)]
        with m.If(vga_tem.tv_out):

            m.d.comb += [
                vga.r.eq(7),
                vga.g.eq(7),
                vga.b.eq(7)]
        with m.Else():
            m.d.comb += [
                vga.r.eq(0),
                vga.g.eq(0),
                vga.b.eq(0)]
        
        return m

platform.build(VGA_test(), do_program=True)

```

I was unable to build and lash the code. It always return 
```
Traceback (most recent call last):
  File "vga_test.py", line 42, in <module>
    platform.build(VGA_test(), do_program=True)
  File "/home/gal3li0/.local/lib/python3.6/site-packages/nmigen/build/plat.py", line 73, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "/home/gal3li0/.local/lib/python3.6/site-packages/nmigen/build/plat.py", line 113, in prepare
    fragment = Fragment.get(elaboratable, self)
  File "/home/gal3li0/.local/lib/python3.6/site-packages/nmigen/hdl/ir.py", line 67, in get
    obj = obj.elaborate(platform)
TypeError: elaborate() takes 1 positional argument but 2 were given

```

<p>
Secondly, as it can be seen in the project repository, I had to copy the resources folder in the nmigen_board repository (https://github.com/m-labs/nmigen-boards/tree/master/nmigen_boards/resources) into my project folder to make it import successfully. Is there a way to do that without copying the folder?</p>
Thanks. 
Regards,
Taiwo

whitequark, 2020-03-12:
Regarding the exception: instead of:
```
def elaborate(self):
```
use:
```
def elaborate(self, platform):
```
Regarding the resources: if you're using a board definition that is not located in `nmigen_boards`, you should use `from nmigen_boards.resources import *` rather than `from .resources import *.

teezzan, 2020-03-12:
Thank you. It worked but still failed to compile. This time, I got a lot of compilation error that seems to stem from the generated verilog.
```
Info: *******************************************************************
Info: Running Quartus Prime Analysis & Synthesis
    Info: Version 19.1.0 Build 670 09/22/2019 SJ Lite Edition
    Info: Copyright (C) 2019  Intel Corporation. All rights reserved.
    Info: Your use of Intel Corporation's design tools, logic functions 
    Info: and other software and tools, and any partner logic 
    Info: functions, and any output files from any of the foregoing 
    Info: (including device programming or simulation files), and any 
    Info: associated documentation or information are expressly subject 
    Info: to the terms and conditions of the Intel Program License 
    Info: Subscription Agreement, the Intel Quartus Prime License Agreement,
    Info: the Intel FPGA IP License Agreement, or other applicable license
    Info: agreement, including, without limitation, that your use is for
    Info: the sole purpose of programming logic devices manufactured by
    Info: Intel and sold by Intel or its authorized distributors.  Please
    Info: refer to the applicable agreement for further details, at
    Info: https://fpgasoftware.intel.com/eula.
    Info: Processing started: Thu Mar 12 19:51:11 2020
Info: Command: quartus_map --rev=top top
Warning (18236): Number of processors has not been specified which may cause overloading on shared machines.  Set the global assignment NUM_PARALLEL_PROCESSORS in your QSF to an appropriate value for best performance.
Info (20030): Parallel compilation is enabled and will use 2 of the 2 processors detected
Error (10170): Verilog HDL syntax error at top.v(148) near text: "=";  expecting an identifier. Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number. File: /home/gal3li0/Documents/electronic/vga nmgen/build/top.v Line: 148
Error (10112): Ignored design unit "U$$0$1" at top.v(120) due to previous errors File: /home/gal3li0/Documents/electronic/vga nmgen/build/top.v Line: 120
Error (10112): Ignored design unit "cd_sync" at top.v(152) due to previous errors File: /home/gal3li0/Documents/electronic/vga nmgen/build/top.v Line: 152
Error (10112): Ignored design unit "reset_sync" at top.v(164) due to previous errors File: /home/gal3li0/Documents/electronic/vga nmgen/build/top.v Line: 164
Error (10112): Ignored design unit "top" at top.v(183) due to previous errors File: /home/gal3li0/Documents/electronic/vga nmgen/build/top.v Line: 183
Info (12021): Found 0 design units, including 0 entities, in source file top.v
Error: Quartus Prime Analysis & Synthesis was unsuccessful. 5 errors, 1 warning
    Error: Peak virtual memory: 873 megabytes
    Error: Processing ended: Thu Mar 12 19:51:51 2020
    Error: Elapsed time: 00:00:40
    Error: Total CPU time (on all processors): 00:00:29
Traceback (most recent call last):
  File "vga_test.py", line 39, in <module>
    platform.build(VGA_test(), do_program=True)
  File "/home/gal3li0/.local/lib/python3.6/site-packages/nmigen/build/plat.py", line 77, in build
    products = plan.execute_local(build_dir)
  File "/home/gal3li0/.local/lib/python3.6/site-packages/nmigen/build/run.py", line 98, in execute_local
    subprocess.check_call(["sh", "{}.sh".format(self.script)])
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 3.

```
What do I do? 
Thanks.

whitequark, 2020-03-12:
Please archive and upload the build directory here, and also the complete source code of the project.

teezzan, 2020-03-12:
Here is the complete source code...
[vga nmgen.zip](https://github.com/m-labs/nmigen/files/4326388/vga.nmgen.zip)

teezzan, 2020-03-12:
I don't understand what you mean by build directory.

whitequark, 2020-03-12:
The bug you're hitting has been fixed in the master branch of nmigen. Please switch to master and try again.

teezzan, 2020-03-12:
Perhaps I should just reinstall. I dont recall switching to another branch of nmigen

whitequark, 2020-03-12:
That should be https://github.com/nmigen/nmigen (the repository where active development happens), not this one.

teezzan, 2020-03-12:
Thanks a lot. It compiles and flash now. 
I am quite glad even though my VGA code did not work.
Thanks once again

closed 2020-03-12 by teezzan