m-labs/nmigen #321: Internal Osciallator Usage ICE40 UP5K

jchidley, 2020-03-15:
Hi there, I own an Upduino 2.1 and Tinyfpga bx.  How do I use the internal oscillators on these boards?  I can see that in Migen the upduino v1 board was supported.

whitequark, 2020-03-15:
Try something like:
```
    m = Module()
    m.domains.sync = sync = ClockDomain()
    m.submodules.osc = Instance("SB_HFOSC",
        p_CLKHF_DIV="0b00", # 48 MHz
        o_CLKHF=sync.clk
    )
```

jchidley, 2020-03-15:
It's not helping that I am a complete beginner at this.  I am getting this error:

`ERROR: Unable to place cell 'osc_OSC' of type 'ICESTORM_HFOSC'`

Which I really hope means something to you.

whitequark, 2020-03-15:
Are you perhaps trying to use it on TinyFPGA BX? The FPGA on that board does not have oscillators.

jchidley, 2020-03-15:
This is what I have right now.

```python
from nmigen import *
import os
import subprocess

from nmigen.build import *
from nmigen.vendor.lattice_ice40 import *
from nmigen_boards.resources import *

class board(LatticeICE40Platform):
    device      = "ICE40UP5K"
    package     = "SG48"
    resources   = [
        *LEDResources(pins="41", attrs=Attrs(IO_STANDARD="SB_LVCMOS")),
    ]

    connectors  = []

    def toolchain_program(self, products, name):
        iceprog = os.environ.get("ICEPROG", "iceprog")
        with products.extract("{}.bin".format(name)) as bitstream_filename:
            subprocess.check_call([iceprog, "-S", bitstream_filename])

class Blinky(Elaboratable):
    def elaborate(self, platform):
        led   = platform.request("led", 0)
        timer = Signal(20)

        m = Module()
        m.domains.sync = sync = ClockDomain()
        m.submodules.osc = Instance(
            "SB_HFOSC",
            p_CLKHF_DIV="0b00",
            o_CLKHF=sync.clk,
        )

        m.d.sync += timer.eq(timer + 1)
        m.d.comb += led.o.eq(timer[-1])
        return m

if __name__ == "__main__":

    platform = board()
    platform.build(Blinky(), do_program=False)
```

whitequark, 2020-03-17:
There is a typo here. This code:

> ```python
>     device      = "ICE40UP5K"
> ```

should be:

> ```python
>     device      = "iCE40UP5K"
> ```

This counts as a usability bug, feel free to open an issue over at https://github.com/nmigen/nmigen so that no one else hits it in the future.

jchidley, 2020-03-17:
Thank you!  Fixing the typo fixed the fixed the `ERROR: Unable to place cell 'osc_OSC' of type 'ICESTORM_HFOSC'`.  I had to read the email on my phone a couple of times before I noticed the incorrect case at the beginning of "iCE40UP5K"!

I still can't get blinky.py to work properly on the Upduino 2.1 board.  That is down to my inexperience with both FPGAs and Python: at least I can control that bit.  I can't close this issue yet but you might feel it is solved. 

I might have more luck if I base the board definition on the `FomuPVTPlatform` one.  My new Fomu should arrive in a month or so, so I can test that too. 

I might switch back to the TinyFPGA BX in the meantime until I have learnt more.

jchidley, 2020-03-21:
This PR https://github.com/nmigen/nmigen/pull/338 fixes this.

closed 2020-04-07 by jchidley