m-labs/nmigen #323: Installation fails if wheel not installed

alanvgreen, 2020-04-03:
Working in a fresh venv environment, nmigen fails to install without first installing `wheel`.

Here is a transcript showing failure to install nmigen and then success after pip installing wheel.

~~~~
(life) osboxes@osboxes:~/life$ pip install git+https://github.com/m-labs/nmigen.git
Collecting git+https://github.com/m-labs/nmigen.git
  Cloning https://github.com/m-labs/nmigen.git to /tmp/pip-req-build-3ep6l8l0
Requirement already satisfied (use --upgrade to upgrade): nmigen==0.2.dev69+g8f5a253 from git+https://github.com/m-labs/nmigen.git in ./lib/python3.7/site-packages
Requirement already satisfied: Jinja2 in ./lib/python3.7/site-packages (from nmigen==0.2.dev69+g8f5a253) (2.11.1)
Requirement already satisfied: pyvcd~=0.1.4 in ./lib/python3.7/site-packages (from nmigen==0.2.dev69+g8f5a253) (0.1.7)
Requirement already satisfied: setuptools in ./lib/python3.7/site-packages (from nmigen==0.2.dev69+g8f5a253) (40.8.0)
Requirement already satisfied: MarkupSafe>=0.23 in ./lib/python3.7/site-packages (from Jinja2->nmigen==0.2.dev69+g8f5a253) (1.1.1)
Requirement already satisfied: six in ./lib/python3.7/site-packages (from pyvcd~=0.1.4->nmigen==0.2.dev69+g8f5a253) (1.14.0)
Building wheels for collected packages: nmigen
  Running setup.py bdist_wheel for nmigen ... error
  Complete output from command /home/osboxes/life/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-3ep6l8l0/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-slvl3t2i --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for nmigen
  Running setup.py clean for nmigen
Failed to build nmigen
(life) osboxes@osboxes:~/life$ pip install wheel
Collecting wheel
  Using cached https://files.pythonhosted.org/packages/8c/23/848298cccf8e40f5bbb59009b32848a4c38f4e7f3364297ab3c3e2e2cd14/wheel-0.34.2-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.34.2
(life) osboxes@osboxes:~/life$ pip install git+https://github.com/m-labs/nmigen.git
Collecting git+https://github.com/m-labs/nmigen.git
  Cloning https://github.com/m-labs/nmigen.git to /tmp/pip-req-build-gmyg1wxw
Requirement already satisfied (use --upgrade to upgrade): nmigen==0.2.dev69+g8f5a253 from git+https://github.com/m-labs/nmigen.git in ./lib/python3.7/site-packages
Requirement already satisfied: Jinja2 in ./lib/python3.7/site-packages (from nmigen==0.2.dev69+g8f5a253) (2.11.1)
Requirement already satisfied: pyvcd~=0.1.4 in ./lib/python3.7/site-packages (from nmigen==0.2.dev69+g8f5a253) (0.1.7)
Requirement already satisfied: setuptools in ./lib/python3.7/site-packages (from nmigen==0.2.dev69+g8f5a253) (40.8.0)
Requirement already satisfied: MarkupSafe>=0.23 in ./lib/python3.7/site-packages (from Jinja2->nmigen==0.2.dev69+g8f5a253) (1.1.1)
Requirement already satisfied: six in ./lib/python3.7/site-packages (from pyvcd~=0.1.4->nmigen==0.2.dev69+g8f5a253) (1.14.0)
Building wheels for collected packages: nmigen
  Running setup.py bdist_wheel for nmigen ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-div1dvh9/wheels/f0/c9/44/bfd7ac2e366c7aac0ef24261738d7ece300c033982b26ca038
Successfully built nmigen

~~~~

My environment: 

````
(life) osboxes@osboxes:~/life$ uname -a
Linux osboxes 5.3.0-45-generic #37-Ubuntu SMP Thu Mar 26 20:41:27 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
(life) osboxes@osboxes:~/life$ python --version
Python 3.7.5
(life) osboxes@osboxes:~/life$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan
````

sbourdeauducq, 2020-04-04:
How did you set up the virtualenv? ``wheel`` is automatically installed when creating a new venv with ``virtualenv dest_dir`` here.

sbourdeauducq, 2020-04-04:
Does adding ``wheel`` to ``setup_requires`` fix the problem for you?

alanvgreen, 2020-04-05:
This is using venv rather than virtualenv. I set it up with:

python3 -m venv <dir>

I'll try to get some hobby-time tonight to test whether modifying setup_requires in setup.py fixes the problem. (Apologies that I didn't try that already - I've been away from Python for a long time and am rusty.)