m-labs/nmigen #341: Python simulator hangs or throws when trying to drive reset from testbench

hansfbaier, 2020-12-25:
I just learned that nmigen handles reset automatically, so in a simple module, I pulled out my own reset logic and tried 
to use the reset logic from the clock domain. That does not seem to be a problem for verilog generation,
but for the python simulator I tried two approaches, and both failed:
1. I get the resetsignal with ResetSignal() and then try to drive it with in the tests. This should work, because the constructors all default to the "sync" domain. But here I get the error:
```
File "/home/jack/riscv/litex-root/nmigen/nmigen/sim/pysim.py", line 209, in __init__
    self.curr = self.next = signal.reset
AttributeError: 'ResetSignal' object has no attribute 'reset'
```
2. I explicitly create a clock domain and feed it into the simulator, but then the simulator does not seem to terminate
[reset-sim-hangs.tar.gz](https://github.com/m-labs/nmigen/files/5741285/reset-sim-hangs.tar.gz)

bvernoux, 2020-12-25:
See issue https://github.com/m-labs/nmigen/issues/342 m-labs/nmigen repository is totally obsolete and active ones is https://github.com/nmigen => https://github.com/nmigen/nmigen ...

hansfbaier, 2020-12-25:
Oops, wrong repository. This one is obsolete.

closed 2020-12-25 by hansfbaier