m-labs/nmigen #342: Please add details that this repository is obsolete and official ones is https://github.com/nmigen

bvernoux, 2020-12-25:
Please add details that this repository is obsolete and official ones is https://github.com/nmigen
m-labs/nmigen is clearly obsolete and seems not updated anymore since few years
Also m-labs does not have nmigen-boards which is mandatory to support easily latest boards ...
Please clarify that as it is a real mess to use this repository which is not updated and does not support correctly latest platform like ECP5 ...
Official maintained ones is https://github.com/nmigen => https://github.com/nmigen/nmigen https://github.com/nmigen/nmigen-boards ...

sbourdeauducq, 2020-12-25:
> m-labs/nmigen is clearly obsolete

No it's not, and you don't know the real story. Please open an issue in nmigen/nmigen and tell whitequark and the other members of team asshole to rename their fork, which is what they should have done in the first place, and to stop being such ungrateful pricks.

closed 2020-12-25 by sbourdeauducq

bvernoux, 2020-12-25:
I do not know the story behind It was just a comment done as a new comer in migen/n-migen world to use correctly ECP5 FPGA with those amazing open source tools migen/n-migen (with Yosys/nextpnr ...)
I hope in future open source community will be less fragmented (and will clarify things without hurting anyone the aim is to push such amazing open source tools to more and more new comer/users/developers...).

sbourdeauducq, 2020-12-25:
> with those amazing open source tools migen/n-migen (with Yosys/nextpnr ...)

There are a number of technical and managerial things in nMigen that I want to do differently, and I am planning to do so at some point (in fact, it will be quite a major rewrite), but I currently have more important things to do. I also have little appetite for interacting further with some of the people around the fork (see below).

> I hope in future open source community will be less fragmented 

There's been indeed a lot of balkanization with open FPGA projects for the past ~10 years and I'm afraid it's not getting any better. The technical quality has been going up though.

As far as the nMigen fork is concerned, that's 100% a problem with whitequark and her friends; it's not me who engages in bullying, mobbing, defamation, other forms of lying, and other ridiculous behavior - while hypocritically advocating for codes of conduct - whenever I don't have things my way.